pybamm-team/liionpack #8: Event handling

TomTranter, 2021-09-29:
Currently there is none except monitoring the voltage limits on all cells manually and stopping the simulation. Would be amazing if we can leverage some of the event handling in PyBaMM

TomTranter, 2021-11-10:
@tinosulzer @rtimms @brosaplanella what are your thoughts on this? I was thinking of converting the event flags to variables and having them get calculated at each time step and output by the liionpack solver like the variables are. We could handle them simply by monitoring for changes without going back and recalculating the solution to find the exact stopping point.

brosaplanella, 2021-11-10:
I am not sure about the insides of liionpack, but aren't the events of the cell models inherited? I mean, if we run SPMs, don't they inherit the events of the SPM? Or are we talking about experiments?

rtimms, 2021-11-10:
So I think the way liionpack is set up (just using the casadi integrator, not the pybamm `step`) it _might_ fail when you hit an event, depending on the model. In this case I guess the whole thing just crashes? 

Even if it doesn't fail it won't return the event time. So some of the models will have stepped past the event. This might be ok if you were just running, say, a constant current discharge since you can just plot the results and see it hit the voltage, but for an experiment, you will get errors from not switching between steps at the right time. 

I think what you are suggesting is kind of similar to how we handle events in casadi in pybamm. You take a global step e.g. 10 seconds and return the results (assuming the solver didn't fail). Then you check the sign of the events, if they have changed you go back and interpolate the solution to find the event time (e.g. hit V_max after 5 seconds) and truncate the solution here. In liionpack you would then pass this solution back to the global solve, being careful about the fact the step size is different here (5s vs 10s).

In pybamm we go back and solve again with a more dense `t_eval` to find the event time more accurately, but we could skip this for now. But we don't do the whole solve again, just the "window" in which the event occurred. So we might have done the global step of 10s with 1s steps in the model solver and the event happened somewhere between 4s and 5s, so we go back and solve from 4s to 5s with a dense grid to find a precise `t_event`.

I think there are 3 sensible options:
1. Parallelise the `step` method in pybamm and let this handle events. I think you would want to use the "safe without grid" mode to avoid recreating the integrator lots of times. Not sure how much work this would be.
2. Handle events in liionpack in a similar way to how we handle events in the cadsadi solver in pybamm
3. Restrict liionpack to ODE models (for now) and just use `scipy` which can do the event handling for you for the individual cells, then just do some work to patch this together with the global steps for the pack

TomTranter, 2021-11-11:
> 1. Parallelise the `step` method in pybamm and let this handle events. I think you would want to use the "safe without grid" mode to avoid recreating the integrator lots of times. Not sure how much work this would be.

I'm beginning to favour this approach

TomTranter, 2021-11-11:
I don't want to end up re-writing a lot of functionality that's already in pybamm but also there will be liionpack specific optimisation that will take place probably. Also I think there's going to be events that break the solver and events that are more like experiment stop conditions.

TomTranter, 2021-11-11:
I'm less concerned about going to the exact event time as our step sizes are small anyway

TomTranter, 2021-11-24:
done in pr #88

closed 2021-11-24 by TomTranter