pybamm-team/liionpack #15: module 'skspatial.objects' not found

PippaVossen, 2021-10-04:
Hi @TomTranter, you use a module named 'skspatial.objects' in the liionpack package - where can I get this from? Is it linked to scipy? I can't seem to find this module anywhere...
Many thanks in advance!

TomTranter, 2021-10-04:
it's the scikit-spatial package. scikit-X packages often abbreviate the module names. There is a requirements.txt file in this package which details all the dependencies

TomTranter, 2021-10-04:
By the way if you navigate a command prompt to the folder you have downloaded or cloned locally you can install liionpack and all dependencies automatically with pip install -e .

TomTranter, 2021-10-04:
Ah my apologies, I had not updated the setup.py file to reflect all the dependencies. It should be there now

closed 2021-10-04 by TomTranter