pybamm-team/liionpack #57: Different Initial condition for the cells in pack

ksnvikrant, 2021-11-03:
I tried to populate different initial electrolyte concentration across the cells in the pack. I have an array of initial electrolyte concentration (initial_eleConc) similar to total heat transfer coefficient (htc). The changes I made to **utils.py**, **simulations.py**, **solver_utils.py** are given below.

Changed line 220, and added line 229 in **utils.py**. 
```python
# -*- coding: utf-8 -*-
"""
Created on Thu Sep 23 10:33:13 2021

@author: Tom
"""
from scipy.interpolate import interp1d, interp2d
import numpy as np
import pandas as pd
import os
import liionpack
from skspatial.objects import Plane
from skspatial.objects import Points

ROOT_DIR = os.path.dirname(os.path.abspath(liionpack.__file__))
CIRCUIT_DIR = os.path.join(ROOT_DIR, "circuits")
DATA_DIR = os.path.join(ROOT_DIR, "data")
INIT_FUNCS = os.path.join(ROOT_DIR, "init_funcs")

def interp_current(df):
    r"""
    Returns an interpolation function for current w.r.t time

    Parameters
    ----------
    df : pandas.DataFrame or Dict
        Contains data for 'Time' and 'Cells Total Current' from which to
        construct an interpolant function

    Returns
    -------
    f : function
        interpolant function of total cell current with time.

    """
    t = df["Time"]
    I = df["Cells Total Current"]
    f = interp1d(t, I)
    return f

def _z_from_plane(X, Y, plane):
    r"""
    Given X and Y and a plane provide Z
    X - temperature
    Y - flow rate
    Z - heat transfer coefficient

    Parameters
    ----------
    X : float (array)
        x-coordinate.
    Y : float (array)
        z-coordinate.
    plane : skspatial.object.Plane
        plane returned from read_cfd_data.

    Returns
    -------
    z : float (array)
        z-coordinate.

    """
    a, b, c = plane.normal
    d = plane.point.dot(plane.normal)
    z = (d - a * X - b * Y) / c
    return z
[... 592 lines cut ...]
    period="10 seconds",
)

# PyBaMM parameters
chemistry = pybamm.parameter_sets.Chen2020
parameter_values = pybamm.ParameterValues(chemistry=chemistry)

# Solve pack
output = lp.solve(netlist=netlist,
                  parameter_values=parameter_values,
                  experiment=experiment,
                  output_variables=output_variables,
                  htc=htc,initial_eleConc=initial_eleConc)
lp.plot_output(output)
```
However, in the simulation the first value of initial electrolyte concentration, i.e., initial_eleConc[0] is populating across all the cells. Here is the output showing that
[image]
:
I think the issue is in line 256 of solver_utils.py, i.e.
```python
integrator, variables_fn, t_eval = _create_casadi_objects(
        I_init, htc[0],initial_eleConc[0], sim, dt, Nspm, nproc, variable_names
    )
```
where it is only taking the first element of the array initial_eleConc[0].  I gave  initial_eleConc array similar to htc(heat transfer coefficient) array. As htc array is going into **lumped** model, so every cell can have their own individual heat transfer coefficient. In contrast, the parameters in the base model seems not populating across the cells.

tinosulzer, 2021-11-03:
Haven't looked specifically into what's happening here, but we should come up with a user interface that allows users to change any parameter like this, instead of adding each one as another argument  to all the functions

TomTranter, 2021-11-05:
> Haven't looked specifically into what's happening here, but we should come up with a user interface that allows users to change any parameter like this, instead of adding each one as another argument to all the functions

Agreed. The current has to be an input for Liionpack to work and is changed on every step. HTC technically doesn't but you would want it to be if solving a dynamic thermal problem and again might change a little bit for every step. The rest are more like ensemble constants and would be different to start with but then not change. Is it worth making this distinction and calling it something else?

ksnvikrant, 2021-11-05:
> Can you make a branch from main with the code changes you made?

I created a branch name **issue-57-init_cond** with the above changes

TomTranter, 2021-11-19:
Hi @ksnvikrant this issue is also related to being able to pass in the model flexibly. I think the answer to both is to set up the simulation and then pass that into the liionpack solvers. Only problem there is how to enforce the current to be an input. I guess throwing errors if it's not might be ok for now. The code changes you made will now be out of date after the solver refactor so I think it's probably easier just to make them again off a new branch. I can combine this with the DFN model issue into a new branch

TomTranter, 2021-11-24:
@ksnvikrant I've just pushed a bunch of changes which should help you with this issue. There is a new notebook on using inputs. Let me know if you have any questions

ksnvikrant, 2021-11-25:
> 
Hi @TomTranter Thanks for the changes. I tried to create an example for different initial SOC condition for the cells in pack. However, it is still taking the first initial SOC of the cell and populating across all other cells. In order to run the example below, first you need to **comment line 44 of solvers.py**, i.e., **#lp.update_init_conc(self.simulation, SoC=initial_soc)**. The example code is here:
```python
import liionpack as lp
import pybamm
import numpy as np
I_mag = 5.0
OCV_init = 4.0  # used for intial guess
Ri_init = 5e-2  # used for intial guess
R_busbar = 1.5e-6  # very small to simulate evenly distributed currents
R_connection = 1e-2
Np = 4
Ns = 1
Nbatt = Np * Ns
netlist = lp.setup_circuit(
    Np=Np, Ns=Ns, Rb=R_busbar, Rc=R_connection, Ri=Ri_init, V=OCV_init, I=I_mag
)

experiment = pybamm.Experiment(
    ["Discharge at 5 A for 30 minutes", "Rest for 15 minutes"],
    period="10 seconds",
)

output_variables = [  
    'X-averaged negative particle surface concentration [mol.m-3]',
    'X-averaged positive particle surface concentration [mol.m-3]',
    ]

chemistry = pybamm.parameter_sets.Chen2020
parameter_values = pybamm.ParameterValues(chemistry=chemistry)
socInit=np.array([0.2, 0.3, 0.4, 0.5])

c_n_max = parameter_values["Maximum concentration in negative electrode [mol.m-3]"]
c_p_max = parameter_values["Maximum concentration in positive electrode [mol.m-3]"]

neg_init=np.zeros(Np*Ns)
pos_init=np.zeros(Np*Ns)

for i in range(Np*Ns):
    x, y = pybamm.lithium_ion.get_initial_stoichiometries(socInit[i], parameter_values)
    neg_init[i], pos_init[i] = x * c_n_max, y * c_p_max

parameter_values.update({"Initial concentration in negative electrode [mol.m-3]": "[input]"})
parameter_values.update({"Initial concentration in positive electrode [mol.m-3]": "[input]"})
inputs = {"Initial concentration in negative electrode [mol.m-3]": neg_init,"Initial concentration in positive electrode [mol.m-3]": pos_init}

print (np.unique(neg_init))

output = lp.solve(
    netlist=netlist,
    parameter_values=parameter_values,
    experiment=experiment,
    inputs=inputs,output_variables=output_variables
)
lp.plot_output(output)
lp.show_plots()

```
The output for surface concentration of negative and positive particles for the cells should look different as the initial SOC of cells are different. However, output is same for all the cells. 
[image]
[image]

TomTranter, 2021-11-26:
We could change the logic in the solvers to not use the initial_soc argument if is None and change the default and that should allow you to customize in this way or we could develop the logic to accept an array of initial_soc

TomTranter, 2021-12-02:
This issue should now be addressed by PR #103 which is merged. See https://github.com/pybamm-team/liionpack/blob/main/examples/different_initial_soc.py for example and let me know if we can close this issue

TomTranter, 2021-12-02:
@ksnvikrant also that other branch is redundant now so can you delete it?

closed 2021-12-03 by TomTranter