pybamm-team/liionpack #71: Allow user to pass in simulation instead of parameter_values

TomTranter, 2021-11-19:
This would probably help with issues #57 and #48 but might cause some confusion. The reason simulation was created behind the scenes was to setup the model with the thermal option and also to make the current and heat transfer coefficients inputs. Actually the thermal model doesn't need to be set by default and the inputs should be handled more generically allowing the user to specify anything as an input. The only think that needs to be an input is the current and even then that may be a sub-case for a more generic ensemble solve which we might want to allow for. For now liionpack does packs so this is probably a feature we can add (without too much trouble I expect)

TomTranter, 2021-11-19:
@tinosulzer @rtimms what do you think about how to handle always setting current as input but allow user to pass simulation. It would sometimes require re-building model I guess

rtimms, 2021-11-19:
Yeah it deffo gives more control to the user in a neatly packaged way (they get to pick options, mesh etc.) this way. Where would the simulation need rebuilding? You can just build once like we do now I think? 

You could always add some checks on options we don’t want to support yet or on what inputs you can pass. 

If not we should at least allow the user to pass a model and any inputs they want, but I can’t see an obvious reason off the top of my head why passing a simulation would be a problem. I guess we need to use the casadi solver to generate the mapper functions or has that changed now?

TomTranter, 2021-11-19:
I was thinking the simulation would need rebuilding if they had not configured current as an input

rtimms, 2021-11-22:
The simulation doesn't build until you call `build` or `solve` (which calls `build`), so you could update the parameters still.

TomTranter, 2021-11-22:
This is causing issues in multiple places. Firstly dask can't serialize the simulation and secondly I need to override the experiment otherwise calls to sim.solve to initialize the solution for the liionpack step fail and it's getting messy. I think it might be easier to stick with what we have for now

closed 2021-11-22 by TomTranter