pybamm-team/liionpack #74: GitHub Action workflow to check conda environment

wigging, 2021-11-19:
Can someone add a GitHub Action to build and check the conda environment? I don't have permission to update the workflow in this repository but I believe it would be something like this:

```yaml
  build:
    needs: style
    runs-on: ubuntu-latest

    steps:
    - uses: actions/checkout@v2
    - name: Set up Python 3.9
      uses: actions/setup-python@v2
      with:
        python-version: 3.9
    - name: Install dependencies
      run: |
        $CONDA/bin/conda env update --file environment.yml --name base
```

wigging, 2021-11-19:
Looks like @Saransh-cpp created the current workflow so maybe he can add a conda build.

Saransh-cpp, 2021-11-19:
@wigging I pushed a new branch that adds conda to GitHub Actions, I'll make a PR if the tests pass.

Saransh-cpp, 2021-11-19:
Oh wait, are we adding the conda build, or are we replacing the old CI with just the conda build? (Right now I am replacing it)

wigging, 2021-11-19:
Add it to what you did before. The idea is to make sure liionpack builds correctly using the pip approach and that it also builds correctly using the conda approach.

PR #75 (2021-11-20): Use Conda in GitHub Actions
Fixes #74 

## Changes/Additions
- CI now also install dependencies and runs tests in a conda environment
- flake8's max-line-length changed to 89 (as it is in the .flake8 file)
- Coverage report is now generated using conda
- Minor style fixes
closed 2021-11-20 by TomTranter