pybamm-team/liionpack #82: Array split does not result in equal division

wigging, 2021-11-22:
Splitting the arrays for attributes `self.split_index` and `self.htc` in solvers.py throws an error if the number of cells is not evenly divisible by the number of processes `nproc`. For example, using `Np = 32`, `Ns = 20` with `nproc = 6` does not work for the Dask and Ray solvers. This can be avoided using `np.array_split` but this creates another error about a singular matrix caused by divide by zero in netlist_utils.py for `g[R_map] = 1 / value[R_map]`.

TomTranter, 2021-11-22:
@wigging  I am aware that my method was crude and was concentrating on just getting parallelism working. I don't know why the R_map would have zeros in it.... Do you wanna try fixing this?

wigging, 2021-11-22:
I started a **fix-arraysplit** branch and made the changes from NumPy `split` to `array_split`. I have not fixed the issue this causes in `netlist_utils.py` but here is the error I'm getting:
```
/liionpack/liionpack/netlist_utils.py:438: RuntimeWarning: divide by zero encountered in true_divide
  g[R_map] = 1 / value[R_map]  # conductance = 1 / R
/opt/miniconda3/envs/liionpack/lib/python3.9/site-packages/scipy/sparse/linalg/dsolve/linsolve.py:206: MatrixRankWarning: Matrix is exactly singular
  warn("Matrix is exactly singular", MatrixRankWarning)
Low voltage limit reached
```

TomTranter, 2021-11-22:
What script are you running?

wigging, 2021-11-22:
Running `big_circuit_ray.py` for `nproc=5` using the Dask or Ray manager.

TomTranter, 2021-11-22:
spm_per_worker needs to be the length of the chunk whereas it's a constant at the moment

wigging, 2021-11-22:
I think I have it fixed. Let me run some examples before pushing up the changes.

closed 2021-11-23 by TomTranter