pybamm-team/liionpack #85: Include circuits, data, init_funcs in setup.py

priyanshuone6, 2021-11-23:
## Describe the bug
Installing liionpack using `pip install -q git+https://github.com/pybamm-team/liionpack.git@main` does not include circuits, data, init_funcs folders in the package

[image]

[image]

## To Reproduce
Steps to reproduce the behaviour:
Run `02 Simulating a bespoke pack using an LTSpice netlist and custom heat transfer.ipynb` from examples/notebooks

## TO DO
Include these folders in setup.py

For Reference:
https://github.com/pybamm-team/PyBaMM/blob/5fbfec8a46945e99106cc166c283201513066425/setup.py#L136-L160

priyanshuone6, 2021-11-23:
@TomTranter @wigging what files and extensions from these folders should I include in the package?

TomTranter, 2021-11-23:
We might want to move all of these to a features folder. I'm not sure what to do with init_funcs as it was used for setting intial SoC which we changed how to do but actually for a pack problem where you might want to intialize with a bunch of different values it could be useful. The other two are quite bespoke and used for an example of loading in your own netlist. For now include everything

PR #89 (2021-11-24): Add circuits, data, init_funcs to pkg
Add circuits, data, init_funcs to pkg
Fixes #85
closed 2021-11-24 by TomTranter