pybamm-team/liionpack #86: ray multiprocessing test on github actions

TomTranter, 2021-11-23:
See PR for failing test

Getting this error running ray on github actions server. Could be something to do with how we're pip installing? @priyanshuone6 any ideas?
RuntimeError: The actor with name ray_actor failed to import on the worker. This may be because needed library dependencies are not installed in the worker environment

priyanshuone6, 2021-11-23:
The same happened when I was working on running jax tests in the workflow, and it got solved after installing it in the tox environment.
Perhaps we need to install and run all these in the relevant conda environment
https://github.com/pybamm-team/liionpack/blob/6db7aa0ff6fabd25111b2dce17c8934f807d3a54/.github/workflows/test_on_push.yml#L50-L57

just like these
https://github.com/pybamm-team/liionpack/blob/6db7aa0ff6fabd25111b2dce17c8934f807d3a54/.github/workflows/test_on_push.yml#L58-L63

priyanshuone6, 2021-11-23:
I can see a commit https://github.com/pybamm-team/liionpack/commit/f4cfbc66747b6fd68d1aa1cee487b0d89fdcee8c#diff-2fbc801faaed9a628483d63edb74d7c5cc8af99f1a311dd46d18e5fe198fbcac by @Saransh-cpp on this, don't know what went wrong

Saransh-cpp, 2021-11-23:
Right now the tests are running twice -

1. Using no environment (using `pip`, including installing all the dependencies using `pip`)
2. Completely in the `conda` environment (including installing all the dependencies in the `conda` environment)

The final error that the workflows show is -
```
ModuleNotFoundError: No module named 'liionpack'
```
when the tests are running using pip, which is weird because there are no issues with other dependencies.

TomTranter, 2021-11-23:
Fixed with pr #87

closed 2021-11-23 by TomTranter