pybamm-team/liionpack #106: Update examples to use latest version

wigging, 2021-12-03:
I pulled down the latest version of liionpack and I'm getting voltage limit warnings for many of the examples. And some of the examples fail to run. Does the test suite check these examples when the main branch is updated or when pull requests are submitted?

TomTranter, 2021-12-07:
notebooks are checked but scripts not. I've been using that scripts folder as a bit of a sandpit to be honest. I think we should get rid of it and just use the notebooks

TomTranter, 2021-12-07:
We can keep them if you like but I use them for longer simulations and to include them in tests would be a pain.

wigging, 2021-12-07:
I would prefer to keep them in the repo. Others may find them useful too. I will go through the script files today and update them for the latest version.

TomTranter, 2021-12-07:
I think the issue is probably the Chen2020 parameter set has initial conditions for particle concentrations near SoC of 1.0 and we start the cycles by charging. initial_soc is also None by default which means it just takes the parameter values defaults. Setting initial_soc to 0.5 fixes this

closed 2021-12-14 by TomTranter