pybamm-team/liionpack #122: Use of the period parameter for an experiment

wigging, 2022-01-11:
In liionpack, the value of the `period` parameter for an experiment is used to determine the length of each time step. In PyBaMM the `period` is how often outputs are recorded (see the [PyBaMM docs]) and therefore is not the time step used for calculations. Unless I'm misunderstanding the documentation, the `period` in liionpack is different than the `period` in PyBaMM. If this is correct, I suggest changing `period` in liionpack to be named something like `timestep` to avoid confusion with PyBaMM. 

[PyBaMM docs]: https://pybamm.readthedocs.io/en/latest/source/experiments/experiment.html

TomTranter, 2022-01-12:
Yes you are correct but this is because the solvers are very different and I didn't want to re-invent the experiment class for liionpack. To be honest I don't think it's really an issue. Having said that I now think that we might get issues when experiments are defined with steps of different periods so maybe it is worth making our own one...

wigging, 2022-01-12:
I see in the liionpack docs that the `period` parameter is defined as the time step. So I guess it's fine as long as it's mentioned in the documentation.

closed 2022-01-12 by wigging