pybamm-team/liionpack #123: power charging

kenan991, 2022-01-12:
Is there a specific reason why other than in PyBaMM there are only constant current operations are supported? in another word is there any way I can dis/charge power from/to battery?
and is liionpack a suitable library for simulating the behavior of a bigger battery like a power bank?

TomTranter, 2022-01-12:
No specific reason other than it hasn't been developed yet. liionpack is still pre-release. If you are interested in adding that functionality we can help and that would be great

TomTranter, 2023-02-01:
@cyasing

TomTranter, 2023-02-01:
So I think we'd need to control power at the pack level and then work out the total pack voltage then divide power by this to get current and apply the single step currents as normal

TomTranter, 2023-02-01:
Have to allow power control in protocols 
https://github.com/pybamm-team/liionpack/blob/develop/liionpack/protocols.py

TomTranter, 2023-02-01:
It's used in the solver here https://github.com/pybamm-team/liionpack/blob/b1a1ddbc6f607f24b3f916b8cc68f4a17980cf2e/liionpack/solvers.py#L319

TomTranter, 2023-02-01:
And here
https://github.com/pybamm-team/liionpack/blob/b1a1ddbc6f607f24b3f916b8cc68f4a17980cf2e/liionpack/solvers.py#L319

TomTranter, 2023-02-01:
And here
https://github.com/pybamm-team/liionpack/blob/b1a1ddbc6f607f24b3f916b8cc68f4a17980cf2e/liionpack/solvers.py#L425

TomTranter, 2023-02-01:
Terminal voltage can be used to divide power to set terminal current. This is set here
https://github.com/pybamm-team/liionpack/blob/b1a1ddbc6f607f24b3f916b8cc68f4a17980cf2e/liionpack/solvers.py#L431
I guess using the value from previous time step should work

cyasing, 2023-02-01:
> @cyasing

Thank you.

cyasing, 2023-02-02:
> Terminal voltage can be used to divide power to set terminal current. This is set here
> 
> https://github.com/pybamm-team/liionpack/blob/b1a1ddbc6f607f24b3f916b8cc68f4a17980cf2e/liionpack/solvers.py#L431
> 
> 
> I guess using the value from previous time step should work

It looks like the terminal voltage is set after solving `solve_circuit_vectorized(netlist)` in the function `_step()`, which in itself requires current value `I_map` to solve: https://github.com/pybamm-team/liionpack/blob/b1a1ddbc6f607f24b3f916b8cc68f4a17980cf2e/liionpack/solvers.py#L425
Power needs to be converted to current before that, doesn't it?

TomTranter, 2023-02-02:
Yep you could loop in the initialisation step to get the right current

OlliRuokojoki, 2023-02-02:
@cyasing I could work with you on this issue. Can I find you for example in the pybamm Slack channel so we could chat?

cyasing, 2023-02-02:
> @cyasing I could work with you on this issue. Can I find you for example in the pybamm Slack channel so we could chat?

@OlliRuokojoki I don't have a slack account yet. I'll ask for an invitation.

TomTranter, 2023-02-02:
Use this invite https://join.slack.com/t/pybamm/shared_invite/zt-1obrjh8fr-FKm~sT8693cZnAcim5dh~A

cyasing, 2023-02-02:
> Use this invite https://join.slack.com/t/pybamm/shared_invite/zt-1obrjh8fr-FKm~sT8693cZnAcim5dh~A

Thanks!

TomTranter, 2023-02-14:
Did you get something working for this? I ask because the script you use on the other issue uses power

cyasing, 2023-02-14:
> Did you get something working for this? I ask because the script you use on the other issue uses power

I have added some code to convert power to current in the way you described, but getting that to test requires me to figure out the problem with the shifting solver, or at least the reason behind the capacity warning coming up in it. It's in a loop.

TomTranter, 2023-02-14:
Is everything you are working on in a branch somewhere I can take a look at?

cyasing, 2023-02-14:
I had forked the repository, actually. It is here: https://github.com/cyasing/liionpack-power-trial.git . Do tell me if forking like this is not something I should do.

OlliRuokojoki, 2023-05-29:
Hi, 
Is anyone still working on this? I'd really need this feature.

TomTranter, 2023-06-08:
FYI - there is a PyBaMM change to experiments coming in to the next release imminently so might make sense to wait for that https://github.com/pybamm-team/PyBaMM/pull/2960

OlliRuokojoki, 2023-06-12:
Hi @TomTranter,
I have now simplified my large battery pack to just a single large battery cell as basic PyBaMM works much better. I may just keep working on my simulation with the single cell to make the process a lot easier. 

By the way does Liionpack support drive cycles? If I used drive cycles and current data I could get the results I'm looking for. At some lower and upper voltages I could stop the drive cycle, write new current data based on the new terminal voltage and then continue the drive cycle at the point it was interrupted by my voltage check. Do you think this approach is feasible with Liionpack?

Thanks for the help!

TomTranter, 2023-06-13:
It does support drive cycles but stopping and re-starting simulations with old solution is not totally straight forward. Our plan is to merge liionpack into PyBaMM so the solvers and experiments will all work exactly the same in future. It's just quite a big job and other things have taken priority right now. If you can get the single PyBaMM model working for your needs I would stick with that

closed 2024-06-26 by TomTranter