pybamm-team/liionpack #126: JOSS review: documentation

mefuller, 2022-01-25:
I am unable to find a link to proper documentation, just example notebooks.
Where is the documentation to be found? Please make the location more visible on the main page

priyanshuone6, 2022-01-25:
@mefuller thanks for catching that. The link to the documentation can be found in the [contributing guidelines](https://github.com/pybamm-team/liionpack/blob/main/docs/contributing.md). I'll make a PR and include it in the readme.

Documentation link - https://liionpack.readthedocs.io/

mefuller, 2022-01-25:
Thanks - I still don't see a clear overview of commands, just examples.
The documentation would definitely need to include that to pass JOSS review

TomTranter, 2022-01-25:
The docs can be found by clicking on the docs badge in the readme on the homepage. I can also include an explicit link

TomTranter, 2022-01-25:
Overview of commands can be found here https://liionpack.readthedocs.io/en/main/reference/

closed 2022-02-02 by mefuller