pybamm-team/liionpack #128: JOSS Review: Paper

mefuller, 2022-02-02:
### Description

Paper needs references updated with DOIs where possible (see https://github.com/openjournals/joss-reviews/issues/4051#issuecomment-1010721400)

Also, please add some "literature review" or survey to identify what other software packages (if any) exist that provide similar capabilities and how these differ.

Thank you

TomTranter, 2022-02-04:
@mefuller I have made a pull request to address these points but I can't assign you to review it. Shall I just merge it in?

mefuller, 2022-02-04:
@TomTranter sure thing - I looked at the PR and the changes are satisfactory

closed 2022-02-04 by mefuller