pybamm-team/liionpack #130: [Bug]: can't conda install via environment.yml on mac

tinosulzer, 2022-02-08:
### liionpack Version

n/a

### Python Version

3.9

### Describe the bug

Install command suggested in the readme fails
```
$ conda env create -f environment.yml
Collecting package metadata (repodata.json): done
Solving environment: failed

ResolvePackageNotFound: 
  - redis
 ```

### Expected behaviour

Installs

### Additional context

Mac OS on M1

Possibly related to https://stackoverflow.com/questions/58219956/how-to-fix-resolvepackagenotfound-error-when-creating-conda-environment

wigging, 2022-02-09:
This may be an issue with installing on the new Apple Silicon M1 Mac. Is Anaconda/Conda supported on M1 Macs? I have an Intel MacBook Pro and I don't have any liionpack installation issues with conda install using the `environment.yml` file.

tinosulzer, 2022-02-09:
I was able to install without any problems by first creating an empty environment then pip installing into it. Have never tried doing it with an `environment.yml` though, maybe that isn't supported. Also I have to use miniforge, don't know if that makes a difference

wigging, 2022-02-09:
Did you create the empty environment using conda or pip?

tinosulzer, 2022-02-09:
conda

wigging, 2022-02-09:
Can you try to install liionpack using the pip venv approach on your M1 Mac and see if you have any issues?

```
# Clone the repository
$ git clone https://github.com/pybamm-team/liionpack.git

# Create a virtual environment in the repository directory
$ cd liionpack
$ python -m venv .venv

# Activate the virtual environment and upgrade pip if venv installed an old version
$ source .venv/bin/activate
$ pip install --upgrade pip

# Install the required packages
$ pip install -r requirements.txt

# Install the liionpack package from within the repository
$ pip install -e .
```

tinosulzer, 2022-02-09:
Yeah that works

wigging, 2022-02-09:
So this looks like a conda issue with M1 Macs and not an issue with liionpack itself. Also, I don't see an Anaconda installer for M1 Macs (https://www.anaconda.com/products/individual). I see a Miniconda installer for the M1 Mac that was released in November for Python 3.8.

tinosulzer, 2022-02-09:
Ok, thanks for following up

closed 2022-02-09 by tinosulzer