pybamm-team/liionpack #131: [Bug]: Package dependency error in paper_example.py

EricaEgg, 2022-02-08:
### liionpack Version

0.2.1

### Python Version

3.9.7

### Describe the bug

I have imported the environment.yml file given in the repo, and successfully built the environment. However, when I run some of the example files, I get a common error. For example, in the paper_example.py script, after running `lp.draw_circuit(netlist, cpt_size=1.0, node_spacing=2.2)`, I get the following error: `RuntimeError: pdflatex is not installed`.

I am using a MacBook with M1 chip, which I know can lead to package dependency problems. Please let me know if I can provide any more info.

wigging, 2022-02-09:
Do you have LaTeX installed on your Mac? See https://tug.org/mactex/ for download and installation instructions.

EricaEgg, 2022-02-09:
Ah, got it! It is resolved

TomTranter, 2022-02-10:
Thanks for spotting this @EricaEgg we'll add some more installation instructions

closed 2022-02-10 by TomTranter