pybamm-team/liionpack #136: Pass `model` to solve as an alternative to `sim_func`

tinosulzer, 2022-02-09:
### Description

Add a `model` keyword argument to `lp.solve` where a user can pass in a pybamm model (instead of via `sim_func`). This is slightly easier than having to write an entire `sim_func` function, if all you want to change is the model

### Motivation

Pro: easier to change the model and leave everything else the same
Con: more ways of doing the same thing

### Possible Implementation

```python
lp.solve(
    netlist=netlist,
    parameter_values=param,
    experiment=experiment,
    initial_soc=initial_soc,
    model=model,
)
```

TomTranter, 2022-02-10:
The reason I have set it up as a sim_func is that the function gets run on every sub-process which is less expensive and memory intensive than passing objects to each sub-process. I agree it's not as easy as passing a model and probably for small problems would work fine but for large scale problems I was getting a lot of issues making pybamm objects and passing them around.

tinosulzer, 2022-02-10:
I was thinking that if the user passes a model instead of a sim_func it would automatically create a basic sim_func using that model. Something like
```python
def get_basic_simulation_function(model=None):
    # Create the pybamm model
    if model is None:
        model = pybamm.lithium_ion.SPM()
        
    def basic_simulation(parameter_values=None):
        """
        Create a Basic PyBaMM simulation set up for integration with liionpack

        Args:
            parameter_values (pybamm.ParameterValues):
                The default is None.

        Returns:
            pybamm.Simulation:
                A simulation that can be solved individually or passed into the
                liionpack solve method

        """
        # Add events to the model
        model = lp.add_events_to_model(model)

        # Set up parameter values
        if parameter_values is None:
            param = pybamm.pybamm.ParameterValues("Chen2020")
        else:
            param = parameter_values.copy()

        # Set up solver and simulation
        solver = pybamm.CasadiSolver(mode="safe")
        sim = pybamm.Simulation(
            model=model,
            parameter_values=param,
            solver=solver,
        )
        return sim

    return basic_simulation

```

TomTranter, 2022-02-11:
@tinosulzer we can set it up and see what the speed is like for a large problem, maybe we could actually get away with passing an unbuilt simulation