pybamm-team/liionpack #137: Symbolic pre-solve for circuit equations?

tinosulzer, 2022-02-10:
Since the structure of the Modified Nodal Analysis equations never changes for a given circuit, we make the circuit set-up and solve time almost negligible by pre-solving the linear system with symbolic values for the resistances and OCVs. I think sympy can do this. Then at each iteration, we just substitute in the resistance and OCV and evaluate the pre-solved equations.

Thinking further along this line of thought, we could (in theory) completely remove the separate pybamm solve <-> model solve by integrating the pre-solved circuit equations into the pybamm model and just solving one big model. However, it's less clear how to parallelize this.

TomTranter, 2022-02-10:
Great ideas, I had thought about making one big system but you do need the stepping for the thermal equations if you include external sources and cell to cell connections that don't exist in the electrical network. I'm very open to exploring these ideas though as if you could just a big system at sundials and get parallelisation done for you that might be better. Definitely we should implement a pre-solve

tinosulzer, 2022-05-24:
Not for a while