pybamm-team/liionpack #144: Post-processing of the data

survik19, 2022-02-25:
Is there a way to save the output of extensive cycle data of battery pack and use it later on in post-processing to generate required plots? Sometimes rerunning the simulation is time-consuming; therefore, it will be helpful to save the output and use it for post-processing.

wigging, 2022-02-25:
@survik19 I created a pull request that may satisfy this issue. I noticed that saving to CSV takes up a lot of disk space compared to using NumPy's binary formats `.npy` and `.npz`.

TomTranter, 2022-02-26:
I think we also need to journal the simulation objects themselves so that you can restart a simulation from a later point if it fails or for other reasons. This will be more tricky I expect and probably not using the npz files but in some ways they might be better than a pickle file which would rely on persisting the class definitions. Something to discuss

closed 2022-05-23 by TomTranter

SyarifulAzis1018, 2022-05-23:
i'm sorry, i try to save data to csv, but if i write the code
lp.save_to_csv(output)
so, the result is
module 'liionpack' has no attribute 'save_to_csv'

SyarifulAzis1018, 2022-05-23:
i'm sorry
i try to save file to csv but the result is 
module 'liionpack' has no attribute 'save_to_csv'

TomTranter, 2022-05-23:
@SyarifulAzis1018 did you pull down the latest main branch and restart your kernel? 
See the save_ouput.py example script for how to use

SyarifulAzis1018, 2022-05-23:
> @SyarifulAzis1018 did you pull down the latest main branch and restart your kernel? See the save_ouput.py example script for how to use

I don't know i use latest main branch or not, i think i use it, because i just installed it. And i use script from example, but the result still same, 
Can you help me? Please

TomTranter, 2022-05-24:
Ok so main branch is slightly behind pip. I'm going to push a new version to pip today

SyarifulAzis1018, 2022-05-24:
> Ok so main branch is slightly behind pip. I'm going to push a new version to pip today

Thank you, my problem is already solved, thank you so much