pybamm-team/liionpack #147: [Bug]: Build failing 'SPM' object has no attribute 'init_eval'

TomTranter, 2022-04-08:
### liionpack Version

Latest

### Python Version

All

### Describe the bug

Current build failing due to expected behaviour from PyBaMM being changed.

SyarifulAzis1018, 2022-04-18:
how can i fixed it?

TomTranter, 2022-04-19:
@SyarifulAzis1018 the problem should be fixed if you are running the latest version of PyBaMM and liionpack from github. You need to clone liionpack and use that instead of pip until the next release. Few more things I need to do before releasing. Or roll back PyBaMM to an earlier release from pip and use that