pybamm-team/liionpack #149: [Feature Request] Saving and replacing or starting with different simulation solutions

Mykle99, 2022-04-09:
### Description

A method to simulate differently aged cells in one and the same circuit with liionpack. After a given amount of cycles one can specify cells in the circuit that are to be removed and replaced with unaged or other cells from another circuit which has been cycled differently.

### Motivation

- Simulate the behaviour of systems which combine aged/non-aged cells
- Simulate the behaviour of systems which contains differently aged cells 
- Expand the number of cells in a circuit. How does behaviour change when the dimentions `Np` x `Ns` are changed? (Either by decreasing or increasing circuit size [I: Which cells should be removed? II: System behaviour with the inclusion of non-aged cells or differently aged cells combined with the initial system])
- In BESS some cells has to be replaced before the system is decommissioned, this can be modelled to simulate behaviour when some cells are replaced with fresh (non-aged) cells (or differently aged cells from another system) whilst the majority of the original cells are left as they are in the BESS 
- Capacity is lost over time and a BESS may need to replace its capacity to sustain its purpuse or expand its capacity for a different use case where more capacity is needed. Simulating this may bring insight into how this can be done efficiently?

### Possible Implementation

Initial condition expansion:
- `different_initial_soc.py` from the liionpack repo. shows how different `SoC` can be set with `update_init_conc(param, SoC=None, update=True)`
- `different_initial_soc.py` could be expanded to include a variable that describes aging such as `SoH` or `RuL`

A general implementation which allows saving and replacing or starting with different simulation solutions:
- Run a cell degradation model over a number of cycles and then copy the solution (which can be different for each specified cell) to initialize the pack
- Chose which cell(s) to replace by defining which position(s) in the `Np` x `Ns` circuit are to be replaced and which cells should remain
- Chose a new `Np` x `Ns` dimention and which cells it contains (either new cells or cells which have been aged)

TomTranter, 2022-04-11:
Thanks @Mykle99 I think a first step would be to change liionpack to accept an initial PyBaMM solution to start from. Currently the solutions are initialised behind the scenes to save transferring data to parallel processors. PyBaMM does not allow you to specify an SoH or RuL as an initial condition because it would be way too complicated and subjective. However what you could do is run a single cell simulation for a number of cycles and copy solutions at the end of different cycles into the pack format. Or like you say, run a pack simulation for a bit then replace some cells with fresh cells. Both methods would require passing in starting solutions rather than starting concentrations.

Mykle99, 2022-04-19:
Is there a way to bounty issues/feature requests?

TomTranter, 2022-04-19:
Do you mean upvote or are you asking if you can pay for it?

Mykle99, 2022-04-19:
Is paying within the guidelines of GitHub, or is it not allowed? From the nature of your answer I assume, there are no personal reputation bounties similar to those found on StackOverflow.

TomTranter, 2022-04-20:
No I don't think so