pybamm-team/liionpack #150: [Bug]: pip install in Windows

brosaplanella, 2022-04-12:
### liionpack Version

v0.3

### Python Version

3.7.7

### Describe the bug

When trying to `pip install liionpack` on Windows it does not work because `ray` can't be pip installed (it needs to use conda, see [here](https://docs.ray.io/en/latest/ray-overview/installation.html#windows-support). Not sure where `ray` is used. If it is not fundamental, can we make it optional (similar to `scikits-ode` solvers in PyBaMM)?

### Steps to Reproduce

1. Get a Windows machine
2. Run `pip install liionpack`
3. See the error

### Expected behaviour

liionpack should be installed without errors.

### Relevant log output

```shell
ERROR: Could not find a version that satisfies the requirement ray (from liionpack) (from versions: none)
ERROR: No matching distribution found for ray
```
```

wigging, 2022-04-12:
Ray, Casadi, and Dask are managers that can be defined for the solver. See the [solver_utils](https://liionpack.readthedocs.io/en/latest/reference/#liionpack.solver_utils.solve) documentation. Ray gives the best parallel performance for large pack simulations using more than 64 CPU cores. I have only used it on Linux and macOS but the [Ray documentation](https://docs.ray.io/en/latest/ray-overview/installation.html) says it has beta support for Windows.

brosaplanella, 2022-04-13:
Yes, but from [this link](https://docs.ray.io/en/latest/ray-overview/installation.html#windows-support) it seems that the only way to install is using conda and, at least from my experience, simply doing `pip install ray` does not work. Therefore, running `pip install liionpack` fails to install liionpack in Windows (due to this problem with `ray`).

TomTranter, 2022-05-24:
@brosaplanella I have no issues using ray on windows and from the link above it can be installed with pip, they just use conda to create the environment first.

TomTranter, 2022-05-25:
I think there is a dependency conflict (possibly sympy with pybamm and lcapy which should be addressed in next release) which means pybamm is getting rolled back to 0.4.0. redis seems to be in the environment.yml file but not the setup.py or requirements.txt

wigging, 2022-05-25:
Ray needed a redis package for running on macOS.

closed 2022-06-09 by TomTranter