pybamm-team/liionpack #151: [Dask-on-Ray] Potential to try in `solvers.py`?

dynamicwebpaige, 2022-04-20:
### Description

Greetings! 

It looks like `liionpack` is using `dask`, and `ray`; but has not yet tested [Dask-on-Ray](https://docs.ray.io/en/latest/data/dask-on-ray.html) -- which is a drop-in replacement for using distributed Dask, with Ray as a backend. 

Would it make sense to try `enable_dask_on_ray()`, and see if you experience a performance improvement for [`solvers.py`](https://github.com/pybamm-team/liionpack/blob/51f25883b73b0e3f9d577cc57eddc41b86f4b088/liionpack/solvers.py)? For [similar projects](https://www.jennakwon.page/2021/03/benchmarks-dask-distributed-vs-ray-for.html), we've seen up to 13.5x speed-up times (and significant cost reductions).

Happy to answer questions, or to make a PR to test. 😄

wigging, 2022-04-20:
@dynamicwebpaige The first link you provide about Dask on Ray states the following:

> Dask-on-Ray is an ongoing project and is not expected to achieve the same performance as using Ray directly.

So based on that statement I don't see how Dask on Ray would improve anything. I have done several comparisons of the Dask and Ray solvers in liionpack and Ray consistently performs better than Dask. Also, in your second link, the 13.5x speed-up refers to Ray vs Dask Distributed. It doesn't mention that Dask on Ray was used to get that speed-up.

TomTranter, 2022-04-28:
Hi @dynamicwebpaige we had a chat with the dask team as we are using actors to persist the state and it seemed like actors were a bit experimental. If you feel like having a look at our code to see if you can speed it up then more than welcome but probably our most likely route going forward is to just use ray which as @wigging mentioned seems to be more reliable for our use case.

TomTranter, 2022-05-24:
closing this issue as no further responses

closed 2022-05-24 by TomTranter