pybamm-team/liionpack #157: [Bug]: lp make lcapy circuit

SyarifulAzis1018, 2022-05-25:
### liionpack Version

0.3

### Python Version

3.8

### Describe the bug

I try to make a circuit from lcapy to connect to liionpack,
my code is
`import pandas as pd`
`import liionpack as lp`
`net_left = lp.setup_circuit(Np=1, Ns=1, terminals="left")`
`netlist = pd.DataFrame({"desc": ["V1"], "node1": [1], "node2": [5], "value": [10]})`
`aa = pd.DataFrame(data = netlist)`
`lp.make_lcapy_circuit(aa)`
`lp.show_plots()`
but my code is error, i don't realy know about parameters or what must be fill on `lp.make_lcapy_circuit()`
Thank you very much

### Relevant log output

```shell
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python38\lib\site-packages\pandas\core\indexes\base.py", line 3621, in get_loc
    return self._engine.get_loc(casted_key)
  File "pandas\_libs\index.pyx", line 136, in pandas._libs.index.IndexEngine.get_loc
  File "pandas\_libs\index.pyx", line 163, in pandas._libs.index.IndexEngine.get_loc
  File "pandas\_libs\hashtable_class_helper.pxi", line 5198, in pandas._libs.hashtable.PyObjectHashTable.get_item
  File "pandas\_libs\hashtable_class_helper.pxi", line 5206, in pandas._libs.hashtable.PyObjectHashTable.get_item
KeyError: 'node1_x'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "c:\Users\user\Documents\TUGAS AKHIR\print circuit.py", line 9, in <module>
    lp.make_lcapy_circuit(aa)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38\lib\site-packages\liionpack\netlist_utils.py", line 626, in make_lcapy_circuit
    I_xs = [net2[I_map]["node1_x"].values[0], net2[I_map]["node2_x"].values[0]]
  File "C:\Users\user\AppData\Local\Programs\Python\Python38\lib\site-packages\pandas\core\frame.py", line 3505, in __getitem__
    indexer = self.columns.get_loc(key)
  File "C:\Users\user\AppData\Local\Programs\Python\Python38\lib\site-packages\pandas\core\indexes\base.py", line 3623, in get_loc
    raise KeyError(key) from err
KeyError: 'node1_x'
```

TomTranter, 2022-05-25:
It's just for use with netlists made using the setup_circuit function

SyarifulAzis1018, 2022-05-25:
if we made netlists from 
`from lcapy import Circuit`
`netlist = Circuit()`??

TomTranter, 2022-05-25:
I use lcapy for plotting the regular packs, the netlist has the form used by LtSpice etc. with an example here https://github.com/pybamm-team/liionpack/blob/main/examples/load_4p1s.py if you want to use lcapy to generate the circuits then convert to netlist that will need some development but I guess shouldn't be too hard. How familiar with lcapy are you?

SyarifulAzis1018, 2022-05-25:
i just learning about lcapy

TomTranter, 2022-05-25:
So I think we can close this issue or make a longer term feature request

closed 2022-05-25 by SyarifulAzis1018