pybamm-team/liionpack #164: [Bug]: The example code on the website cannot be run

canterville12, 2022-06-13:
### liionpack Version

Most recent version, just after the "pip install liionpack" 

### Python Version

3.9.13

### Describe the bug

When opening the "01 Getting Started" page [here](https://liionpack.readthedocs.io/en/latest/examples/01%20Getting%20Started/), after following the instructions exactly, the code cannot run, it says: "AttributeError: param.C_n_init does not exist. It may have been renamed to param.n.C_init".

tinosulzer, 2022-06-13:
Working on this

tinosulzer, 2022-06-13:
Looks like this has been fixed in main but not yet released

TomTranter, 2022-07-01:
Hi @canterville12 I just published a new release on pypi which hopefully should resolve your issue.

TomTranter, 2022-07-29:
No response so assume fixed

closed 2022-07-29 by TomTranter