pybamm-team/liionpack #167: CONFIGURATION CONNECTION OF THE BATTERY PACK

SyarifulAzis1018, 2022-07-05:
### Description

So far, the circuit connection in liionpack library use series then parallel connection. in the other side, we can setting the connection with the parallel first and then parallel

### Motivation

I read from some literature, this configuration more reliable for unbalance voltage of battery. so if liionpack can make more configuration of battery its so very nice.

### Possible Implementation

i think the implementation can simulate energy storage for power grid.

TomTranter, 2022-07-29:
If you can generate a circuit diagram in LtSpice or something we can import netlist and then we can see how easy to auto generate. Thanks for the suggestion

SyarifulAzis1018, 2022-09-06:
[https://drive.google.com/file/d/1e1DCaLWT3BzXe37hdTaYFMSm03JQk4_x/view?usp=sharing](url)
this file about the circuit 3p2s, thanks

TomTranter, 2022-09-06:
Do you know if the convention is to call this configuration 3p2s and the way liionpack does it 2s3p? I think this should be easy to automate, just wondering about the best way to define the layout with terminology or maybe an argument like series_first = True

SyarifulAzis1018, 2022-09-06:
I think it's goog choices to define the layout

jiean520, 2023-02-13:
> I think it's good choices to define the layout

Do you have a good way to do it? I also face this problem now.

PR #221 (2023-03-03): Option to specify parallel-strings or series-groups
Hi! I ran into #167  recently, and realized there's a pretty easy fix by adding a keyword argument to just add a busbar on every row instead of just the top row. I'm pretty sure in practice most packs use the "series-groups" configuration rather than the "parallel-strings" configuration which is currently implemented in liionpack

Fixes: #167
closed 2023-03-03 by TomTranter