pybamm-team/liionpack #172: Runtime errro: pdflatex is not installed[Bug]:

Sourabh-Jadhav-2, 2022-07-27:
### liionpack Version

0.3.2

### Python Version

3.8

### Describe the bug

I tried simple code from documentation to check whether its working or not. then I got runtime error: pdflatex is not installed. I am using windows

priyanshuone6, 2022-07-27:
Please refer [this](https://lcapy.readthedocs.io/en/latest/install.html) link on how to install pdflatex and required dependencies for windows.

Sourabh-Jadhav-2, 2022-07-27:
> Please refer [this](https://lcapy.readthedocs.io/en/latest/install.html) link on how to install pdflatex and required dependencies for windows.

[image]

TomTranter, 2022-07-29:
You need tex. Please go through the above link carefully

closed 2022-12-13 by TomTranter

balaji3112, 2023-03-14:
I'm facing the same issue. I tried installing the required dependencies in the link but still got the same error.

  File "C:\Users\Welcome\AppData\Local\Programs\Python\Python38\lib\site-packages\lcapy\system.py", line 72, in checkexe
    raise RuntimeError('%s is not installed' % program)
RuntimeError: pdflatex is not installed

TomTranter, 2023-03-14:
Make sure you have the filepath in your environment path

balaji3112, 2023-03-15:
I added the pdflatex path to the environment variables, but the problem persists.

TomTranter, 2023-03-15:
Try adding to the top of the list in the Path variable for both user and system. I had an issue the other day with this myself and it was fixed by moving up the list