pybamm-team/liionpack #174: Experiment[Bug]:

Sourabh-Jadhav-2, 2022-07-28:
### liionpack Version

0.3.2

### Python Version

3.8

### Describe the bug

Why this is not possible?
"Charge at 1C until 4.2V",
"Hold at 4.2V until C/20"

wigging, 2022-07-28:
I think the upper and lower cutoff voltage is 4.2 volts. Try your experiment with 4.1 V or lower.

Sourabh-Jadhav-2, 2022-07-28:
But 4.2V is working finely in pybamm

Sourabh-Jadhav-2, 2022-07-29:
> In liionpack voltage limits are checked as shown below. So if your experiment exceeds these voltage limits then it will exit the simulation or at least show a warning.
> 
> https://github.com/pybamm-team/liionpack/blob/306ef75e2f20bd86ff8541db065e7175ff1a8b1a/liionpack/solvers.py#L441-L448

[image]
[image]
[image]

Sourabh-Jadhav-2, 2022-07-29:
> [image] [image] [image]

I want to implement CCCV in liionpack. I already implemented successfully CCCV in pybamm

TomTranter, 2022-07-29:
Apologies for not checking these issues regularly. I need to sort out my github notifications. As discussed on the PyBaMM discussion this functionality needs to be added. Liionpack only works for CC and drive cycles not voltage or power driven experiments. Just FYI this will probably be quite a significant development and is not a high priority for me right now

wigging, 2022-07-29:
@Sourabh-Jadhav-2 GitHub supports syntax highlighted code blocks. See the links below for more information. There's no need to take screenshots of code or terminal output.

https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code

https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

philgravy, 2024-07-29:
Were you ever able to resolve the issue? I'm experiencing the same problem