pybamm-team/liionpack #183: [Bug]: Benchmarks failing

TomTranter, 2022-08-24:
### liionpack Version

Latest

### Python Version

3.8

### Describe the bug

This was the last benchmark to pass
https://github.com/pybamm-team/liionpack/runs/7624552852?check_suite_focus=true
This was the first one to fail
https://github.com/pybamm-team/liionpack/runs/7644850569?check_suite_focus=true

Failure started to happen after we changed the default branch to be develop I think
@Saransh-cpp any suggestions?

### Relevant log output

```shell
Step that fails:
Run asv machine --machine "GitHubRunner"
· No information stored about machine 'fv-az27[8](https://github.com/pybamm-team/liionpack/runs/7644850569?check_suite_focus=true#step:5:9)-386'. I know about nothing.
  

· Getting existing hashes
· Error running /usr/bin/git rev-list --first-parent main (exit status [12](https://github.com/pybamm-team/liionpack/runs/7644850569?check_suite_focus=true#step:5:13)8)
  STDOUT -------->
  
  STDERR -------->
  fatal: ambiguous argument 'main': unknown revision or path not in the working tree.
  Use '--' to separate paths from revisions, like this:
  'git <command> [<revision>...] -- [<file>...]'
```

wigging, 2022-08-24:
Can you keep the default branch as `main`? You can make the branch protected so people can't directly push/pull to it.

closed 2022-08-30 by TomTranter

Saransh-cpp, 2022-08-30:
> Can you keep the default branch as `main`? You can make the branch protected so people can't directly push/pull to it.

Keeping `main` as the default branch had some loopholes (given that we wanted to test `main` with the latest tagged version of `PyBaMM` and `develop` with the `develop` version of `PyBaMM`). This is explained in much more detail here - #177.

PR #184 (2022-08-30): Fix the failing benchmarking suite
Fixes #183 

Changes the default branch to `develop`!

The suite is running here - https://github.com/pybamm-team/liionpack/runs/8039227245?check_suite_focus=true - and should pass!