pybamm-team/liionpack #203: [Bug]: lp.plot_output(output) and lp.plot_pack provide no output plots when run as a python script

skywo1f, 2022-12-16:
### liionpack Version

June 2022

### Python Version

3.8

### Describe the bug

I am able to somewhat run the examples, but I get no output. I tried adding a line such as lp.plot() or lp.show() after the plot commands, but they are unrecognized.

### Steps to Reproduce

run any basic example as a python script

### Expected behaviour

plotting the results, like in the example notebook

wigging, 2022-12-16:
Which example are you running? Is it one of the examples in [liionpack/examples](https://github.com/pybamm-team/liionpack/tree/develop/examples)? Also, the command to show plots is `lp.show_plots()` not `lp.show()`.

skywo1f, 2022-12-16:
ah, my mistake. I forgot that in a notebook, plots dont require that extra command "show_plots" that you need in a script.

closed 2022-12-16 by skywo1f