pybamm-team/liionpack #204: [Bug]: Known compatible PyBaMM and liionpack versions ?

arnauddevie, 2022-12-30:
### liionpack Version

0.3.2dev (`326b954169ba19a68acd2a8efae542c7f1458f0b` commit)

### Python Version

3.9.15

### Describe the bug

Hi Dev team,

I really appreciate the effort the community is putting in PyBaMM and now liionpack.
I'm eager to run the example notebooks from the repo but I keep running into various runtime errors on multiples OSes and with multiple versions of liionpack (0.3.1, 0.3.2, 0.3.2dev) (see below).

I tried using Colab as a workaround for now but the pip install liionpack cell fails so it's a no-go.

I see bug reports for these issues already and some work has been done apparently suggesting it has been fixed in some commit.

Now, I would like to know if there are 100% known compatible versions of liionpack and PyBaMM that I can start with?

Can someone point me in the right direction please?
I can install either pip versions or specific commits.

Best regards,

### Steps to Reproduce

```python
print(lp.__version__)
print(pybamm.__version__)
print(python_version())
```
```
0.3.2
22.11.1
3.9.15
```

### Expected behaviour

Able to run the 9 examples notebooks at:
https://colab.research.google.com/github/pybamm-team/liionpack/blob/main/

### Relevant log output

```shell
Sample errors:
AttributeError: 'ElectrodeSOHSolver' object has no attribute 'get_initial_stoichiometries'
SolverError: Events ['Maximum voltage'] are non-positive at initial conditions
TypeError: solve() got an unexpected keyword argument 'htc'
KeyError: op['electric']
```

### Additional context

Tried on Windows and Ubuntu 20.04.
Focusing on Ubuntu 20.04 at the moment.

TomTranter, 2023-01-05:
Hi @arnauddevie thanks for your issue and sorry for the inconvenience caused. I am working to publish a new version of the code that should have all the fixes in place will let you know when complete. Hopefully tomorrow. By the way - I'd be interested to know what you are using our code for and if you are interested in finding out more about future projects. I'll send connection on linkedin if that's ok

TomTranter, 2023-01-05:
The release is now on pip

arnauddevie, 2023-01-05:
@TomTranter Thank you very much.
I can confirm I was able to run all 9 examples notebooks without issues with the following configuration:
Ubuntu 22.04.1 LTS (GNU/Linux 5.15.79.1-microsoft-standard-WSL2 x86_64)

``` python
from platform import python_version
print(lp.__version__)
print(pybamm.__version__)
print(python_version())
```
```
0.3.3
22.12
3.9.15
```

closed 2023-01-05 by arnauddevie