pybamm-team/liionpack #208: [Bug]: Update thermal external simulation

TomTranter, 2023-01-05:
### liionpack Version

0.3.2

### Python Version

3.8.3

### Describe the bug

External thermal simulation needs updating to use input instead of external submodel

TomTranter, 2023-01-12:
`import liionpack as lp
import pybamm
import numpy as np
import matplotlib.pyplot as plt
plt.close('all')

def thermal_external(parameter_values=None):
    """
    Create a PyBaMM simulation set up for integration with liionpack.
    External thermal option is used so that temperature dependence can be
    included in models but temperature supplied by another algorithm. This
    is useful for packs and cells where thermal connections are seperate or
    distinct from electrical connections.

    Args:
        parameter_values (pybamm.ParameterValues):
            The default is None.

    Returns:
        pybamm.Simulation:
            A simulation that can be solved individually or passed into the
            liionpack solve method

    """
    # Create the pybamm model
    model = pybamm.lithium_ion.SPMe(
        options={
            # "thermal": "lumped",
            "timescale": 10,
        }
    )

    # Add events to the model
    model = lp.add_events_to_model(model)

    # Set up parameter values
    if parameter_values is None:
        parameter_values = pybamm.ParameterValues("Chen2020")

    # Change the ambient temperature to be an input controlled by the
    # external circuit
    parameter_values["Ambient temperature [K]"] = pybamm.InputParameter(
        "Input temperature [K]"
    )
    parameter_values["Initial temperature [K]"] = pybamm.InputParameter(
        "Input temperature [K]"
    )

    # Set up solver and simulation
    solver = pybamm.CasadiSolver(mode="safe")
    sim = pybamm.Simulation(
        model=model,
        parameter_values=parameter_values,
        solver=solver,
    )
    return sim

# Generate the netlist
netlist = lp.setup_circuit(Np=4, Ns=1, Rb=1e-3, Rc=1e-2)

# Define some additional variables to output
output_variables = [
    "Volume-averaged cell temperature [K]",
    "Volume-averaged total heating [W.m-3]"
]

# Cycling experiment, using PyBaMM
experiment = pybamm.Experiment([
    "Charge at 5 A for 30 minutes"],
    period="10 seconds")

# PyBaMM battery parameters
parameter_values = pybamm.ParameterValues("Chen2020")

# Solve the pack problem
temps = np.ones(4) * 300 + np.arange(4) * 10
inputs = {"Input temperature [K]": temps}
output = lp.solve(netlist=netlist,
                  sim_func=thermal_external,
                  inputs=inputs,
                  parameter_values=parameter_values,
                  experiment=experiment,
                  output_variables=output_variables,
                  initial_soc=0.5)

# Display the results
lp.plot_output(output, color="white")`

TomTranter, 2023-01-12:
@tinosulzer @rtimms this doesn't give the same results as before. Without the lumped thermal model you get a fixed temperature and no heat sources and with the lumped model you get a varying temperature and heat sources. Before you used to get a fixed temperature (supplied externally) and heat sources (to be able to calculate that temperature)

rtimms, 2023-01-12:
Try with 

```python3
model = pybamm.lithium_ion.SPMe(
    options={
        "calculate heat source for isothermal models": "true",
        "cell geometry": "arbitrary",
        "dimensionality": 0,
        "thermal": "isothermal",
        "timescale": 1,
    }
)
```

closed 2023-01-12 by TomTranter