pybamm-team/liionpack #212: Catch "High/Low voltage limit reached"

avarf, 2023-01-11:
### Description

I couldn't find a way to catch the "High/Low voltage limit reached" condition. I checked the solver and it is [logging the message](https://github.com/pybamm-team/liionpack/blob/a1c02b2b1bf54a0e751bdb9123c2c16b8fb5dd58/liionpack/solvers.py#L445) and returns `vlims_ok = False` but it is not passed/visible at the main output and the output is just a dictionary with less number of elements than it should.

It would be great if you can pass this exception/warning explicitly as a field in the output or return two values, one the output and the other one the exception.

### Additional context

Code:
```python
output = lp.solve(
    netlist=netlist,
    parameter_values=parameter_values,
    experiment=experiment,
    initial_soc=SoC,
)
```

TomTranter, 2023-01-12:
Hey @avarf thanks for raising the issue. Do you want to contribute the change?

TomTranter, 2023-04-06:
I have had no response from you so assume this is not a major problem. You are right to identify this as an issue but it is quite minor and should be addressed in future releases once the solving and experiment instruction handling is managed by PyBaMM directly

closed 2023-04-06 by TomTranter