pybamm-team/liionpack #228: [Bug]:always high voltage limit

jiean520, 2023-03-04:
### liionpack Version

0.3.4

### Python Version

3.9

### Describe the bug

The newest version of liionpack 0.3.4 can not run any example code. Always reached high voltage limit. The same code can be run in the version of 0.3.3

TomTranter, 2023-03-04:
Try upgrading pybamm too

jiean520, 2023-03-04:
> Try upgrading pybamm too

Both liionpack(0.3.4) and pybamm(23.2) are newest. Still can not run.

TomTranter, 2023-03-04:
Ok I had some issues with branches yesterday when I made the release but all tests were passing. I'll take a look ASAP

TomTranter, 2023-03-04:
Right sorry about this. Can you roll back to previous version of liionpack for now. I have current version in step with the PyBaMM develop branch.

TomTranter, 2023-04-06:
This should be fixed now in latest versions on pip

closed 2023-04-06 by TomTranter