pybamm-team/liionpack #255: [Bug]: Liionpack not simulating models other than SPM and SPMe

MustafaDragonborne, 2023-04-14:
### liionpack Version

0.3.5

### Python Version

3.9.16

### Describe the bug

I tried to simulate a basic experiment using liionpack for a 1p100s battery pack using the DFN model and the Chen2020 parameter set but the model wouldn't solve. The SPM and SPMe model works though. Using the MPM model throws a KeyError: "'Negative area-weighted particle-size distribution [m-1]' not found. Best matches are ['Negative particle radius [m]', 'Negative electrode diffusivity [m2.s-1]', 'Negative current collector conductivity [S.m-1]']".

Any solutions to work with the DFN model are highly appreciated. 

### Steps to Reproduce

```
from IPython.core.interactiveshell import InteractiveShell
InteractiveShell.ast_node_interactivity = "all"

import time
from datetime import timedelta
import os

import pandas as pd
import numpy as np
import matplotlib.pyplot as plt
from sklearn.metrics import mean_absolute_error, mean_squared_error

import pybamm
from pybamm import ParameterValues
pybamm.set_logging_level('NOTICE')
import liionpack as lp

# set the parameter values according to chemistry
parameter_values_chen = pybamm.ParameterValues('Chen2020')

# create the experiment
experiment = pybamm.Experiment(
    [
        "Charge with 5A for 100 seconds",
        "Rest for 100 seconds",
        "Discharge with 5A for 100 seconds",        
    ] * 1,
    period='0.5 second'
)

# Simulation function
def mysim_func(parameter_values):
    # create the model
    model = pybamm.lithium_ion.DFN()

    # Set up solver and simulation
    # solver = pybamm.CasadiSolver(mode="safe", dt_max=600)
    solver = pybamm.CasadiSolver(mode="fast")
    sim = pybamm.Simulation(
        model=model,
        parameter_values=parameter_values,
        solver=solver
    )
    return sim

Np = 1
Ns = 100
netlist = lp.setup_circuit(
    Np=Np, Ns=Ns, Ri=1e-10, Rc=1e-10, Rb=1e-10, Rt=1e-10
)

t0 = time.time()
output = lp.solve(
    netlist=netlist,
    sim_func=mysim_func,
    parameter_values=parameter_values_chen,
    experiment=experiment,
    # inputs=inputs,
    initial_soc=0.5
)
t1 = time.time()
f'The simulation has finished running.'
f'Simulation time = {timedelta(seconds=round(t1-t0))}'

# Plot system level current and voltage values
# Plot current values
fig, (ax1, ax2) = plt.subplots(nrows=2, ncols=1, figsize=(16,10), sharex=True)

# Pack current
ax1.plot(output['Pack current [A]'], '-o')
ax1.set_title(f'Pack current [A] configuration {Np}p{Ns}s')
ax1.set_ylabel('[A]')
ax1.grid()

# Pack terminal voltage
ax2.plot(output['Pack terminal voltage [V]'],'-o')
# ax2.plot(output['Terminal voltage [V]'].sum(axis=1), '-*')
# ax2.plot(curr['timestamp'], curr['voltage'])
ax2.set_title(f'Pack terminal voltage [V]')
ax2.set_ylabel('[V]')
ax2.legend(['Liionpack', 'Liionpack sum of cells'])
ax2.grid();
```

### Expected behaviour

[image]

hectormara, 2024-12-16:
Using the DFN model causes a runtime error into CASIDI:

RuntimeError: Error in Function::call for 'F' [IdasInterface] at D:\bld\casadi_1705758725091\work\casadi\core\function.cpp:1401:
Error in Function::call for 'F' [IdasInterface] at D:\bld\casadi_1705758725091\work\casadi\core\function.cpp:330:
D:\bld\casadi_1705758725091\work\casadi\interfaces\sundials\idas_interface.cpp:596: IDACalcIC returned "IDA_LINESEARCH_FAIL". Consult IDAS documentation.

Confirmed that works with SPM and SPMe. But there is a voltage drop that I cannot fix.
[image]

xLeoniex, 2025-06-09:
@Santiagopeacely Try to create a developer version of Liionpack and modify the following lines of code in the solver_utils file so that Liionpack also works with the DFN solver:

In the _serial_step function, change the original line:

zf = casadi_sol["zf"]

to:

zf = casadi.horzcat(z0, casadi_sol["zf"])

And in the _mapped_step function, change:

y_alg = zf[:, start : start + nt]

to:

y_alg = casadi.horzcat(z0[:, i], zf[:, start : start + nt])

Santiagopeacely, 2025-06-09:
@xLeoniex Thank you for your reply and answer. According to your opinion, I tried and found that the above-mentioned error will still appear. As I am not a professional developer, I have no way to start with this error. Please help me answer it again. Thank you.

[image]

[image]